satnogs/satnogs-org #56: Build a RaspberryPi with webcam for streaming GS

comzeradd, 2015-01-13:


acinonyx, 2015-04-28:
Hardware arrived back to hackerspace. Should be functional. Someone needs to test it.

ppapadeas, 2015-06-02:
Marios is working on it

ppapadeas, 2015-10-06:
Pinged Marios again

papachristoumarios, 2015-10-06:
Yes, but I am busy with school + fixing some minor bugs (+ documentating) and many more (related to other projects). I'll inform you upon updates. I have also uploaded some 3D designs for the sensor enclosure under the `dev` branch at http://github.com/papachristoumarios/satnogs-meteo (I should transfer ownership to `satnogs` organization some time, shouldn't I?). You can check the code yourself / yourselves, as well, at the `dev` branch (I have added direct support for I2C / IO between the sensors and RPi).