satnogs/satnogs-org #57: HAM radio association research

ppapadeas, 2015-01-13:


ppapadeas, 2015-02-14:
http://www.yme.gr/?tid=263&aid=0
http://www.yme.gr/imagebank/categories/ctg745_9_1192093566.pdf

Άρθρο 7, παρ. 2 [pdf page 5]:
("...Άδεια ραδιοερασιτεχνικού σταθμού ασυρμάτου λαμ−
βάνουν:...")
"2. Τα μη κερδοσκοπικά σωματεία ή σύλλογοι που πλη−
ρούν τις παρακάτω προϋποθέσεις:
2.1. Έχουν συσταθεί και λειτουργούν νόμιμα.
2.2. Από το καταστατικό τους προκύπτει ότι αποσκο−
πούν στην υποστήριξη και ανάπτυξη της υπηρεσίας
ραδιοερασιτέχνη.
2.3. Υπάρχει απόφαση του σωματείου ή συλλόγου με
την οποία ορίζεται μέλος του, αδειούχος ραδιοερασιτέ−
χνης «κατηγορίας 1», ως ο υπεύθυνος λειτουργίας του
ραδιοερασιτεχνικού σταθμού (club station)."

ppapadeas, 2015-03-10:
Ask legal

comzeradd, 2015-03-16:
Legal gave green light

closed 2015-03-26 by comzeradd