satnogs/satnogs-org #59: Measure LNA4ALL gain

acinonyx, 2015-01-20:


ppapadeas, 2015-03-10:
Attenuators missing

zisi, 2015-04-06:
Measure LNA4All

In order to amplify the weak signal and reduce (as possible) the noise that captured by antenna we use the Low Noise Amplifier(LNA), LNA4all.

The LNA4all is connected in the output of antenna and in the input of receiver (as close as possible).
The properties of LNA4all are:

 Noise Figure: <1dB

 Gain: 11.5 - 23.5 dB

 Frequency range : 28kHz - 2.5GHz

 Power Suply: 6 - 12V or 5V (regulated) (we use 9V)

 The antenna input and output to receiver are matched to 50 ohms

 Low cost

We measure the LNA4all to confirm the above values. We used miniVNA Tiny and the 20dB attenuator.

The output signal (DUT) of minyVNA Tiny is 5.15dB after the LNA4all the signal may be ~25dB, but the maximum measured signal by minyVNA Tiny is 10dBm (10mW).
So we have problem to measure without use attenuator. With above case we decided to use 20dB attenuator.

20dB Attenuator

[image]

In below figures are presented the testing setup of LNA4all with attenuator and miniVNA and the measurements which is exported from the miniVNA.

Testing setup

[image]

[image]

Measurements

[image]

The gain of the LNA4all in ~147MHz is 20dB+3.07dB-5.15dB = 17.92 dB

in ~283MHz is 20dB+3.33dB-5.15dB = 18.18 dB

in ~437MHz is 20dB+2.32dB-5.15dB = 17.17 dB

pad: https://pad.hackerspace.gr/p/LNApost

ppapadeas, 2015-04-14:
r+ ready for publishing.

closed 2015-04-21 by ppapadeas