satnogs/satnogs-org #95: Setup vm1.satnogs.org

ppapadeas, 2015-03-18:
For migration of:

* satnogs.org
* librespacefoundation.org
* community.satnogs.org
* mail.satnogs.org

comzeradd, 2015-04-20:
We should start phasing out the mailing list. No need to migrate mailman.

comzeradd, 2016-11-01:
Things pending yet:

2 Wordpress sites: lsf, satnogs
1 Discoursse: satnogs

comzeradd, 2017-02-20:
Discourse has been migrated. Pinging @axilleas in case he wants to complete the wordpress ansible role.

One more thing not listed here is the Owncloud, that ideally should be migrate to Nextcloud.

axilleas, 2017-02-21:
Did you change anything in https://github.com/satnogs/satnogs-ansible?

comzeradd, 2017-02-21:
Nope. At least nothing related to the Wordpress role. But I haven't tested it either.

comzeradd, 2017-03-28:
DNS is still propagating, but all sites are moved to vm1

closed 2017-03-28 by comzeradd