satnogs/satnogs-org #170: Measure SV3EXP LNA

ppapadeas, 2015-11-28:


ppapadeas, 2015-11-28:
Need to ask how to supply voltage.

ppapadeas, 2015-12-02:
From communications:
Για να λειτουργήσει το LNA θα πρέπει να το τροφοδοτήστε με 5V από τη μεριά της εξόδου (όχι της κεραίας). Αυτό μπορεί να γίνει πολύ εύκολα στο RTL SDR παίρνοντας 5V από το USB και βάζοντας ένα πηνείο (RF choke) ανάμεσα στην είσοδο της κεραίας και τη τροφοδοσία.
Σ'αυτά που σας έχω στείλει δεν εχω βάλει θερμοσυστελόμενο για να φαίνονται τα parts, οπότε δεν μπορεί κανείς να διακρίνει εύκολα ποιά είναι η είσοδος και ποιά η έξοδος του LNA. Στον αρσενικό SMA μπαίνει η κεραία και στο θηλυκό είναι η έξοδος.
Συνημμένο είναι και το σχηματικό.
[image]

papamat, 2015-12-02:
Μ αυτόν τον τρόπο δεν έχεις dc στην είσοδο του receiver σου; Βάζεις δλδ το
rf choke στην πηγή για να μην πάει το rf προς τα εκεί αλλά μήπως χρειάζεται
και ένα dc cutoff στην είσοδο του sdr; Δεν ξέρω αν κάνω λάθος αλλά δεν με
ψήνει να έχω μια dc στην είσοδο.

zisi, 2016-12-06:
According to data-sheet of PSA4-5043:
* Noise Figure: <1.3dB
 * 50-500MHz: 0.65 - 0.73 dB, at 5V power supply
* Gain: 7.2 - 24 dB
 * 50-500MHz: 22.1 - 25.4 dB, at 5V power supply
* Frequency range : 50MHz - 4GHz
* Power Supply: 5V from RF cable with Bias-T

The measurements are done with miniVNA, ~30dB attenuator and with 5V power supply. The attenuator is made with [rf-biscuit](https://rfbiscuit.wordpress.com/author/oggstreamer/) and that is TL plot.

[image]

The Z - impedance isn't exactly 50Ω in all measured bandwidth.

[image]

From below plot, we calculate the gain of LNA.

[image]
 
According to above plots, 
* at ~144.547MHz, - 25.09+29.52 = 4.43dB
* at ~436.367MHZ, -15.14+28.17 = 13.03dB

The power supply is 5V@99mA, after the LDO is 1.5V. The LNA is powered by 1.5V.
I think that, the measurements it isn't suit with data-sheet numbers, because of low power supply.
The [LNA4ALL] (https://lna4all.blogspot.gr/) is designed with the same PA and [here](https://community.satnogs.org/t/measure-the-gain-of-lna4all/262) are the measurements.
Also the insertion loss from Bias-T is approximately 0.05dB.

Any ideas?

closed 2017-01-02 by ppapadeas