satnogs/satnogs-org #183: Buy magnets for encoders

zisi, 2015-12-10:
Diametric Magnet NdFeB, Grade N35, D6x2.5mm
http://ams.com/eng/Products/Position-Sensors/Magnets/AS5000-MD6H-3

ppapadeas, 2016-01-05:
Ordered 3 samples.

ppapadeas, 2016-01-20:
Samples arrived

ppapadeas, 2016-02-09:
Need to buy 60. Pierros to proceed with order.

ppapadeas, 2016-03-01:
Check with Petros on local vendor.

zisi, 2016-03-29:
We buy [this magnet](https://www.magnet-shop.com/neodymium/discmagnets/discmagnet-4.0-x-4.0-mm-n35h-nickel-diametral-120c) for testing.

ppapadeas, 2016-10-18:
Need to do a shopping stroll

zisi, 2016-11-06:
We can buy [these magnets](https://www.aliexpress.com/item/100-pcs-NdFeB-Magnet-Disc-Dia-6x2-5-mm-0-236-Diametrically-Magnetized-Strong-Magnet-Neodymium/32639986878.html?ws_ab_test=searchweb0_0,searchweb201602_2_10091_10090_10088_10089,searchweb201603_1&btsid=30e9cd2c-1650-412b-940c-815c7d456839) for 30 stations.

ppapadeas, 2016-11-07:
Purchase made!

closed 2016-11-07 by ppapadeas