satnogs/satnogs-org #192: 30 Kits Buy List

ppapadeas, 2016-03-01:


zisi, 2016-03-01:
- [ ] T-slots
- [ ] Aluminum Axis Φ40mm
- [ ] Pulleys, Belts, Aluminum Box
- [ ] Wires, crimps, terminals female
- [ ] PCB and parts
- [ ] pro micro (arduino), PSU (voltage regulator)
- [ ] Magnets for encoders
- [ ] Rotator Cover (plexiglass)

ppapadeas, 2016-11-01:
Need to break down to individual issues and close.

zisi, 2016-11-09:
The individual issues are [#203](https://github.com/satnogs/satnogs-org/issues/203),
[#204](https://github.com/satnogs/satnogs-org/issues/204),
[#201](https://github.com/satnogs/satnogs-org/issues/201),
[#202](https://github.com/satnogs/satnogs-org/issues/202),
[#183](https://github.com/satnogs/satnogs-org/issues/183),
[#198](https://github.com/satnogs/satnogs-org/issues/198).

ppapadeas, 2016-11-21:
Closed in favor the issues above.

closed 2016-11-21 by ppapadeas